runeharlyk/SpotMicroESP32-Leika #1: Unable to Test Repository as It Cannot Be Uploaded

robo78, 2023-08-12:
platformio.exe run
Processing esp32cam (board: esp32cam; platform: espressif32; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 (6.3.2) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 47 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP Async WebServer @ 1.2.3+sha.f71e3d4
|-- ArduinoJson @ 6.21.3
|-- Bounce2 @ 2.71.0
|-- NewPing @ 1.9.7
|-- Adafruit SSD1306 @ 2.5.7
|-- Adafruit GFX Library @ 1.11.7
|-- Adafruit BusIO @ 1.14.1
|-- Adafruit PWM Servo Driver Library @ 2.4.1
|-- Adafruit ADS1X15 @ 2.4.0
|-- Adafruit HMC5883 Unified @ 1.2.1
|-- Adafruit Unified Sensor @ 1.1.11
|-- UrlEncode @ 1.0.1
|-- MPU6050_light @ 1.1.0
|-- HTTPClient @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- WiFi @ 2.0.0
|-- Wire @ 2.0.0
|-- Update @ 2.0.0
Building in release mode
Compiling .pio\build\esp32cam\src\camera.cpp.o
Compiling .pio\build\esp32cam\src\deviceconfig.cpp.o
Compiling .pio\build\esp32cam\src\jsonserializer.cpp.o
Compiling .pio\build\esp32cam\src\main.cpp.o
Compiling .pio\build\esp32cam\src\network.cpp.o
Compiling .pio\build\esp32cam\src\servo.cpp.o
Compiling .pio\build\esp32cam\src\webserver.cpp.o
Compiling .pio\build\esp32cam\lib8da\FS\vfs_api.cpp.o
In file included from src/camera.cpp:2:
include/camera_pins.h:296:2: error: #error "Camera model not selected"
 #error "Camera model not selected"
  ^~~~~
In file included from include/globals.h:80,
                 from include/servo.h:5,
                 from src/camera.cpp:1:
include/webserver.h: In constructor 'CWebServer::CWebServer()':
include/webserver.h:123:19: error: 'HTTP_PORT' was not declared in this scope
         : _server(HTTP_PORT)
                   ^~~~~~~~~
include/webserver.h:123:19: note: suggested alternative: 'HTTP_POST'
         : _server(HTTP_PORT)
                   ^~~~~~~~~
                   HTTP_POST
include/webserver.h:125:15: error: 'WEBSOCKET_PATH' was not declared in this scope
         , _ws(WEBSOCKET_PATH)
               ^~~~~~~~~~~~~~
include/webserver.h:125:15: note: suggested alternative: 'LWIP_SOCKET_POLL'
         , _ws(WEBSOCKET_PATH)
               ^~~~~~~~~~~~~~
               LWIP_SOCKET_POLL
include/webserver.h: In member function 'void CWebServer::begin()':
[... 340 lines cut ...]
                   ^~~~~~~~~
                   HTTP_POST
include/webserver.h:125:15: error: 'WEBSOCKET_PATH' was not declared in this scope
         , _ws(WEBSOCKET_PATH)
               ^~~~~~~~~~~~~~
include/webserver.h:125:15: note: suggested alternative: 'LWIP_SOCKET_POLL'
         , _ws(WEBSOCKET_PATH)
               ^~~~~~~~~~~~~~
               LWIP_SOCKET_POLL
include/webserver.h: In member function 'void CWebServer::begin()':
include/webserver.h:133:56: error: 'HOSTNAME' was not declared in this scope
         DefaultHeaders::Instance().addHeader("Server", HOSTNAME);
                                                        ^~~~~~~~
include/webserver.h:133:56: note: suggested alternative: 'ESTALE'
         DefaultHeaders::Instance().addHeader("Server", HOSTNAME);
                                                        ^~~~~~~~
                                                        ESTALE
*** [.pio\build\esp32cam\src\camera.cpp.o] Error 1
*** [.pio\build\esp32cam\src\deviceconfig.cpp.o] Error 1
*** [.pio\build\esp32cam\src\network.cpp.o] Error 1
*** [.pio\build\esp32cam\src\main.cpp.o] Error 1
*** [.pio\build\esp32cam\src\servo.cpp.o] Error 1
*** [.pio\build\esp32cam\src\jsonserializer.cpp.o] Error 1
*** [.pio\build\esp32cam\src\webserver.cpp.o] Error 1
============================================== [FAILED] Took 7.41 seconds ==============================================

runeharlyk, 2023-08-14:
Hi! Thank you for showing interest in my project.
It seems I forgot to add the example configuration and section in the readme about what to do with it.
I have added both now. To build the project for the first time you have to:
1. Make a copy of *`include/secrets.example.h`* in the same folder
1. Rename the copy to *`secrets.h`*
1. Configure relevant settings

Let me know if you get it to work

robo78, 2023-08-15:
Duplicate data is in config.h and secrets.h. The tile is not visible on the network despite these settings.

runeharlyk, 2023-08-15:
I have remove the duplicated defines 😁 
What system are you using? And Is it the device that cant connect to the network or do you mean you cannot see it using the mDNS address?

robo78, 2023-08-15:
npm run dev uruchmia page on localhost but esp32cam does not fire anything and I do not see the connection.I see the IP of the device but I do not know how to connect and on what port, serial monitor does not display any information

robo78, 2023-08-16:
The board was broken, now it works, testing

closed 2023-09-02 by runeharlyk