runeharlyk/SpotMicroESP32-Leika #36: Issue building with PIO

Ian-cmd-ops, 2024-04-22:
|-- Adafruit SSD1306 @ 2.5.9
|-- Adafruit GFX Library @ 1.11.9
|-- Adafruit BusIO @ 1.16.0
|-- Adafruit PWM Servo Driver Library @ 2.4.1
|-- Adafruit ADS1X15 @ 2.5.0
|-- Adafruit HMC5883 Unified @ 1.2.3
|-- Adafruit Unified Sensor @ 1.1.14
|-- UrlEncode @ 1.0.1
|-- MPU6050_light @ 1.1.0
|-- SPI @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- WiFi @ 2.0.0
|-- Wire @ 2.0.0
|-- Update @ 2.0.0
Building in release mode
Compiling .pio\build\esp32cam\src\AsyncJpegStream.cpp.o
Compiling .pio\build\esp32cam\src\camera.cpp.o
Compiling .pio\build\esp32cam\src\deviceconfig.cpp.o
Compiling .pio\build\esp32cam\src\jsonserializer.cpp.o
Compiling .pio\build\esp32cam\src\main.cpp.o
Compiling .pio\build\esp32cam\src\movement.cpp.o
Compiling .pio\build\esp32cam\src\network.cpp.o
Compiling .pio\build\esp32cam\src\servo.cpp.o
Compiling .pio\build\esp32cam\src\webserver.cpp.o
Building .pio\build\esp32cam\bootloader.bin
Generating partitions .pio\build\esp32cam\partitions.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\esp32cam\lib81e\AsyncTCP\AsyncTCP.cpp.o
In file included from c:\users\rip_h\.platformio\packages\toolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\sys\reent.h:503,
                 from C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/sys/reent.h:17,
                 from c:\users\rip_h\.platformio\packages\toolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\stdio.h:60,
                 from C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:27,
                 from .pio/libdeps/esp32cam/ESP Async WebServer/src/ESPAsyncWebServer.h:24,
                 from include/AsyncJpegStream.h:1,
                 from src/AsyncJpegStream.cpp:1:
C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/soc/spinlock.h: In function 'void spinlock_initialize(spinlock_t*)':
C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/soc/spinlock.h:46:5: error: '__assert_func' was not declared in this scope
     assert(lock);
     ^~~~~~
C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/soc/spinlock.h: In function 'bool spinlock_acquire(spinlock_t*, int32_t)':
C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/soc/spinlock.h:75:5: error: '__assert_func' was not declared in this scope
     assert(lock);
     ^~~~~~
C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/soc/spinlock.h: In function 'void spinlock_release(spinlock_t*)':
C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/soc/spinlock.h:154:5: error: '__assert_func' was not declared in this scope
     assert(lock);
     ^~~~~~
Compiling .pio\build\esp32cam\libc46\FS\FS.cpp.o
Compiling .pio\build\esp32cam\libc46\FS\vfs_api.cpp.o
*** [.pio\build\esp32cam\src\AsyncJpegStream.cpp.o] Error 1
In file included from c:\users\rip_h\.platformio\packages\toolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\sys\reent.h:503,
                 from C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/sys/reent.h:17,
                 from c:\users\rip_h\.platformio\packages\toolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\stdio.h:60,
                 from C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:27,
                 from .pio/libdeps/esp32cam/MPU6050_light/src/MPU6050_light.h:16,
                 from src/movement.cpp:3:
C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/soc/spinlock.h: In function 'void spinlock_initialize(spinlock_t*)':
C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/soc/spinlock.h:46:5: error: '__assert_func' was not declared in this scope
     assert(lock);
     ^~~~~~
C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/soc/spinlock.h: In function 'bool spinlock_acquire(spinlock_t*, int32_t)':
C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/soc/spinlock.h:75:5: error: '__assert_func' was not declared in this scope
     assert(lock);
[... 219 lines cut ...]
C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/soc/spinlock.h:154:5: error: '__assert_func' was not declared in this scope
     assert(lock);
     ^~~~~~
*** [.pio\build\esp32cam\libc46\FS\FS.cpp.o] Error 1
*** [.pio\build\esp32cam\libc46\FS\vfs_api.cpp.o] Error 1
src/main.cpp: In function 'void setup()':
C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_common/include/esp_assert.h:39:13: error: '__assert_func' was not declared in this scope
             assert(#MSG && (CONDITION));                                                                    \
             ^~~~~~
C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:96:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^~~~~~~~~~~~~~~~~
C:/Users/rip_h/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:175:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((addr), WRITE_PERI_REG);                                                               \
             ^~~~~~~~~~~~~~~~~~~
src/main.cpp:11:5: note: in expansion of macro 'WRITE_PERI_REG'
     WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);
     ^~~~~~~~~~~~~~
*** [.pio\build\esp32cam\src\camera.cpp.o] Error 1
*** [.pio\build\esp32cam\src\deviceconfig.cpp.o] Error 1
*** [.pio\build\esp32cam\src\network.cpp.o] Error 1
*** [.pio\build\esp32cam\src\main.cpp.o] Error 1
*** [.pio\build\esp32cam\src\jsonserializer.cpp.o] Error 1
*** [.pio\build\esp32cam\src\servo.cpp.o] Error 1
*** [.pio\build\esp32cam\src\webserver.cpp.o] Error 1

runeharlyk, 2024-04-23:
What platform are you building for? And can you build other projects fine?

runeharlyk, 2024-04-23:
I haven't experienced it myself unfortunately, but found this [post](https://community.platformio.org/t/error-when-executing-build-platformio-vscode-windows-7/33639) where someone suggest deleting your `C:/Users/rip_h/.platformio` folder an letting it reinstall itself.

Ian-cmd-ops, 2024-04-25:
> I haven't experienced it myself unfortunately, but found this [post](https://community.platformio.org/t/error-when-executing-build-platformio-vscode-windows-7/33639) where someone suggest deleting your `C:/Users/rip_h/.platformio` folder an letting it reinstall itself.

This solved my problem but can you create steps to troubleshoot it in the how to or create a troubleshooting part? 
Also after it built, I discovered my ESP32cam is bad.

runeharlyk, 2024-04-25:
That good to hear. I'm have planned to make a thorough documentation of the project (#27). I can include a section there.

I will closes this issue and updated the issue regarding a troubleshooting guide

closed 2024-04-25 by runeharlyk