runeharlyk/SpotMicroESP32-Leika #37: ESP32 Cam not working

Ian-cmd-ops, 2024-04-26:
I can't get the my ESP32 cam to stream or view after I got my ESP32 Uploaded with via PIO.

Ian-cmd-ops, 2024-04-26:
[image]
[image]
do I need the other modules attached first before it runs

Ian-cmd-ops, 2024-04-26:
#define USE_PSRAM true

#define USE_WIFI true

#define WAIT_FOR_WIFI false

#define USE_WEBSERVER true

#define USE_WEBSERVER_SSL false

#define USE_WEBSOCKET true

#define USE_OAT false

#define USE_NTP false

#define USE_MDNS true

#define USE_DNS_SERVER false

#define USE_REMOTE_SERIAL false

#define USE_LOW_POWER false

#define USE_CAMERA true

#define USE_MPU true

#define USE_POWER_BUTTON true

#define USE_USS true

runeharlyk, 2024-04-30:
The camera stream should work out of the box. Is there anything in the browser console log?

Ian-cmd-ops, 2024-05-22:
Bad ESP32 Cam

closed 2024-05-22 by Ian-cmd-ops