runeharlyk/SpotMicroESP32-Leika #42: Documentation for New Built

Ian-cmd-ops, 2024-05-22:
>  *  Executing task: C:\Users\rip_h\.platformio\penv\Scripts\platformio.exe run --target buildfs --environment esp32dev 
> 
> Processing esp32dev (board: esp32dev; platform: espressif32 @ 6.6.0; framework: arduino)
> ----------------------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> running: build_app.py
> Newest file: 2024-05-21 21:11:44.673394, output file: 2024-05-21 21:11:46.935157
> SSL Cert Store: Certificate bundle downloaded to: ssl_certs\cacert.pem
> SSL Cert Store: Parsing certificates from ssl_certs\cacert.pem
> SSL Cert Store: Successfully added 48 certificates
> SSL Cert Store: Successfully added 48 certificates in total
> SSL Cert Store: Successfully created src\certs\x509_crt_bundle.bin
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
> PLATFORM: Espressif 32 (6.6.0) > Espressif ESP32 Dev Module
> HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
> DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
> PACKAGES:
>  - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
>  - tool-esptoolpy @ 1.40501.0 (4.5.1)
>  - tool-mklittlefs @ 1.203.210628 (2.3)
>  - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ strict
> Found 45 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- ArduinoJson @ 7.0.4
> |-- PsychicMqttClient @ 0.1.0+sha.1a26bd1
> |-- NewPing @ 1.9.7
> |-- Adafruit SSD1306 @ 2.5.10
> |-- Adafruit GFX Library @ 1.11.9
> |-- Adafruit BusIO @ 1.16.1
> |-- MPU6050_light @ 1.1.0
> |-- SPI @ 2.0.0
> |-- ESP32-sveltekit
> |-- PsychicHttp @ 1.0.1
> Building in release mode
> Building FS image from 'data' directory to .pio\build\esp32dev\littlefs.bin
> warning: can't read source directory
> *** [.pio\build\esp32dev\littlefs.bin] Error 1
> ============================================================ [FAILED] Took 2.95 seconds ============================================================
> 
> Environment    Status    Duration
> -------------  --------  ------------
> esp32dev       FAILED    00:00:02.952
> ====================================================== 1 failed, 0 succeeded in 00:00:02.952 ====================================================== 
> 
>  *  The terminal process "C:\Users\rip_h\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'buildfs', '--environment', 'esp32dev'" terminated with exit code: 1. 
>  *  Terminal will be reused by tasks, press any key to close it. 
> 
>  *  Executing task: C:\Users\rip_h\.platformio\penv\Scripts\platformio.exe run --target buildfs --environment esp32dev 
> 
> Processing esp32dev (board: esp32dev; platform: espressif32 @ 6.6.0; framework: arduino)
> ----------------------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> running: build_app.py
> Newest file: 2024-05-21 21:11:44.673394, output file: 2024-05-21 21:11:46.935157
> SSL Cert Store: Certificate bundle downloaded to: ssl_certs\cacert.pem
> SSL Cert Store: Parsing certificates from ssl_certs\cacert.pem
> SSL Cert Store: Successfully added 48 certificates
> SSL Cert Store: Successfully added 48 certificates in total
> SSL Cert Store: Successfully created src\certs\x509_crt_bundle.bin
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
> PLATFORM: Espressif 32 (6.6.0) > Espressif ESP32 Dev Module
> HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
> DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
> PACKAGES:
>  - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
>  - tool-esptoolpy @ 1.40501.0 (4.5.1)
>  - tool-mklittlefs @ 1.203.210628 (2.3)
>  - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ strict
> Found 45 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- ArduinoJson @ 7.0.4
> |-- PsychicMqttClient @ 0.1.0+sha.1a26bd1
> |-- NewPing @ 1.9.7
> |-- Adafruit SSD1306 @ 2.5.10
> |-- Adafruit GFX Library @ 1.11.9
> |-- Adafruit BusIO @ 1.16.1
> |-- MPU6050_light @ 1.1.0
> |-- SPI @ 2.0.0
> |-- ESP32-sveltekit
> |-- PsychicHttp @ 1.0.1
> Building in release mode
> Building FS image from 'data' directory to .pio\build\esp32dev\littlefs.bin
> warning: can't read source directory
> *** [.pio\build\esp32dev\littlefs.bin] Error 1
> ============================================================ [FAILED] Took 3.00 seconds ============================================================
> 
> Environment    Status    Duration
> -------------  --------  -------------
> esp32dev       FAILED    00:00:02.1000
> ====================================================== 1 failed, 0 succeeded in 00:00:02.1000 ======================================================
> 
>  *  The terminal process "C:\Users\rip_h\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'buildfs', '--environment', 'esp32dev'" terminated with exit code: 1. 
>  *  Terminal will be reused by tasks, press any key to close it

runeharlyk, 2024-05-22:
Thank for notifying me about the problem. I have added a pre build step to ensure the data folder exist. Let me know if it fixes the problem and if you experience other problems

Ian-cmd-ops, 2024-05-23:
[image]

Ian-cmd-ops, 2024-05-23:
Also can you update the documentation for this new version?

Ian-cmd-ops, 2024-05-23:
```

Processing esp32dev (board: esp32dev; platform: espressif32 @ 6.6.0; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
running: build_app.py
Newest file: 2024-05-22 21:53:23.498062, output file: 2024-05-22 22:01:20.297650
SSL Cert Store: Certificate bundle downloaded to: ssl_certs\cacert.pem
SSL Cert Store: Parsing certificates from ssl_certs\cacert.pem
SSL Cert Store: Successfully added 48 certificates
SSL Cert Store: Successfully added 48 certificates in total
SSL Cert Store: Successfully created src\certs\x509_crt_bundle.bin
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.6.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 46 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 7.0.4
|-- PsychicMqttClient @ 0.1.0+sha.1a26bd1
|-- NewPing @ 1.9.7
|-- Adafruit SSD1306 @ 2.5.10
|-- Adafruit GFX Library @ 1.11.9
|-- Adafruit BusIO @ 1.16.1
|-- MPU6050_light @ 1.1.0
|-- SPI @ 2.0.0
|-- ESP32-sveltekit
|-- PsychicHttp @ 1.0.1
Building in release mode
Building FS image from 'data' directory to .pio\build\esp32dev\littlefs.bin
/connections/mqtt.html
/connections/ntp.html
/connections.html
/controller/stream.html
/controller.html
/favicon.ico
/index.html
/logo512.png
/manifest.json
/peripherals/camera.html
/peripherals.html
/settings.html
/spot_micro.urdf.xacro
/stl.zip
/sw.js
/system/filesystem.html
/system/metrics.html
/system/status.html
/system/update.html
/system.html
/user.html
/wifi/ap.html
/wifi/sta.html
/wifi.html
/_app/env.js
/_app/immutable/assets/0.css
/_app/immutable/assets/20.css
/_app/immutable/assets/logo512.png
/_app/immutable/assets/_layout.css
/_app/immutable/assets/_page.css
/_app/immutable/chunks/24-hours.js
/_app/immutable/chunks/alert-triangle.js
/_app/immutable/chunks/analytics.js
/_app/immutable/chunks/api.js
/_app/immutable/chunks/buffer-utilities.js
/_app/immutable/chunks/camera-outline.js
/_app/immutable/chunks/Collapsible.js
/_app/immutable/chunks/compareVersions.js
/_app/immutable/chunks/ConfirmDialog.js
/_app/immutable/chunks/each.js
/_app/immutable/chunks/entry.js
/_app/immutable/chunks/file-service.js
/_app/immutable/chunks/folder-outline.js
/_app/immutable/chunks/home.js
/_app/immutable/chunks/index.js
/_app/immutable/chunks/index2.js
/_app/immutable/chunks/index3.js
/_app/immutable/chunks/InfoDialog.js
/_app/immutable/chunks/InputPassword.js
/_app/immutable/chunks/model-store.js
/_app/immutable/chunks/model-utilities.js
/_app/immutable/chunks/notifications.js
/_app/immutable/chunks/pencil.js
/_app/immutable/chunks/preload-helper.js
/_app/immutable/chunks/reload.js
/_app/immutable/chunks/result.js
/_app/immutable/chunks/RSSIIndicator.js
/_app/immutable/chunks/scheduler.js
/_app/immutable/chunks/SettingsCard.js
/_app/immutable/chunks/socket.js
/_app/immutable/chunks/Spinner.js
/_app/immutable/chunks/spread.js
/_app/immutable/chunks/stores.js
/_app/immutable/chunks/string-utilities.js
/_app/immutable/chunks/topology-star-3.js
/_app/immutable/chunks/Visualization.js
/_app/immutable/chunks/_commonjsHelpers.js
/_app/immutable/entry/app.js
/_app/immutable/entry/start.js
/_app/immutable/nodes/0.js
/_app/immutable/nodes/1.js
/_app/immutable/nodes/10.js
/_app/immutable/nodes/11.js
/_app/immutable/nodes/12.js
/_app/immutable/nodes/13.js
/_app/immutable/nodes/14.js
/_app/immutable/nodes/15.js
/_app/immutable/nodes/16.js
/_app/immutable/nodes/17.js
/_app/immutable/nodes/18.js
/_app/immutable/nodes/19.js
lfs warn:475: No more free space 520
lfs_write error(-28): File system is full.
/_app/version.json

error adding file!
Error for adding content from nodes!
=========================================================== [SUCCESS] Took 2.89 seconds ===========================================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       SUCCESS   00:00:02.887
=========================================================== 1 succeeded in 00:00:02.887 =========================================================== 
 *  Terminal will be reused by tasks, press any key to close it. 
```

made some progress but I keep getting this error

runeharlyk, 2024-05-23:
In the new version the data folder is supposed to be *empty and the app is embedded in the firmware to make OAT easier to manage. 
Are you using the default build settings?

Ian-cmd-ops, 2024-05-23:
[image]
[image]
I just cleared the data folder. How do I compile it with Build or with Build Filesystem?

Ian-cmd-ops, 2024-05-25:
[image]

Ian-cmd-ops, 2024-05-25:
[image]

runeharlyk, 2024-05-25:
Found a problem with the build output and have merged the fix (#44).
You should now be able to upload the newest firmware with the built app

Ian-cmd-ops, 2024-05-25:
How long should it take to build the file?

Ian-cmd-ops, 2024-05-25:
Got the ESP32CAM broadcast the SSID 
[image]

and I'm able to connect when connecting to 192.168.4.1, 
[image]

I get this error.

Ian-cmd-ops, 2024-05-26:
I got it to work but only after using verbose Build and  verbose Upload

runeharlyk, 2024-05-26:
The camera settings and livestream is accessible from `/peripherals/camera`

Ian-cmd-ops, 2024-05-27:
I'll close this issue. But I need to open another one about assembly and pinouts for the esp32cam pinout

closed 2024-05-27 by Ian-cmd-ops