runeharlyk/SpotMicroESP32-Leika #45: Pin Out documentation

Ian-cmd-ops, 2024-05-27:
I'm looking for better documentation for pin out and better electronics schematics/layout for assembly. Might need an additional part for I2C bus expansion.

Ian-cmd-ops, 2024-05-28:
Also /peripherals/cams doesn't work and I can't connect to my local network. Only the AP works 
[image]

runeharlyk, 2024-05-28:
I have added some [documentation](https://github.com/runeharlyk/SpotMicroESP32-Leika/tree/master/docs) as well as a [circuit digram](https://github.com/runeharlyk/SpotMicroESP32-Leika/blob/master/docs/2_assembly.md) on how to get started. Its still a bit rough, but should contain some useful information.

As for you problems, what does the serial monitor say? Does the esp32 try to connect? Have you added your wifi from `/wifi/sta`?

Ian-cmd-ops, 2024-05-29:
I got the wifi to work. My latest issue is that I get this.
Also Peripherals on the main page have a drop down. 
I noticed that you don't have a calibration down or have a settings to PCA9685 for calibration the servos 
[image]

Ian-cmd-ops, 2024-05-31:
> I have added some [documentation](https://github.com/runeharlyk/SpotMicroESP32-Leika/tree/master/docs) as well as a [circuit [...]

i added it and got it the SSID to connect.

Ian-cmd-ops, 2024-06-02:
[image]
Still having issues with the camera on this latest build. Using a ESP32 Cam.

Ian-cmd-ops, 2024-06-03:
Marking issue as closed. Didn’t use the correct build environment.

closed 2024-06-03 by Ian-cmd-ops