runeharlyk/SpotMicroESP32-Leika #53: GPS and Having Second ESP32 for Sensors,

Ian-cmd-ops, 2024-06-10:
I have enjoyed working with you on this project and having your assistance in making this leg of SpotMicro project a reality. I enjoy the challenge of trying to troubleshoot issues. As I was reading the goal you have outlined, a idea dawned on me about have GPS integrations could be done with the use of ESP32 LoRA  modules for sensor management. They have GPS, ESP32, LoRA for future development (a small mesh of Leika could communicate) . If you need assistance, I can make CAD models for addition ESP32 modules. Cheers Mate

runeharlyk, 2024-06-18:
I'm currently looking on how to connect a neo 6m GPS module to the ESP32 cam, but as most pins are already is in use it poses some issues. Would you like to have the ESP32 Lora module be the main controller or talk to another? I have through about splitting the firmware #34 to make the robots more customizable.

Ian-cmd-ops, 2024-06-22:
An ESP32 LoRA module for controller would be cool. I could look into having the work together and navigate, If it's easier and simpler for you.

Ian-cmd-ops, 2024-06-26:
I think creating a platform to build off would be great. Controlling the robot via the ESP32 have it future capabilities for controlled Pi  control having an ARM and LIDAR would be nice.