runeharlyk/SpotMicroESP32-Leika #58: No I2C communication along with Connectivity Issues

Ian-cmd-ops, 2024-06-19:
```
--- Terminal on COM8 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx�x␀�x�x�x�����xx����␀��x␀���xx

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46

rst:ets Jul 29 2019 12:21:46
[... 124 lines cut ...]
[  3331][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  4332][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  5333][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  6334][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  7436][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  8540][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  8541][E][IMUService.h:35] begin(): [IMUService] MPU initialize failed
[  9542][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 10543][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 11544][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 12545][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 12546][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[ 13558][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 13565][D][EventSocket.cpp:203] onSubscribe(): [EventSocket] onSubscribe for event: servoConfiguration
[ 14596][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 15603][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 15609][I][ServoController.h:169] configure(): [ServoController] Configured with oscillator frequency 27000000 and PWM frequency 50
[ 15611][I][APSettingsService.cpp:89] startAP(): [APSettingsService] Starting software access point
[ 15626][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 15631][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[ 16204][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 11 - AP_STOP
[ 16205][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[ 16208][I][APSettingsService.cpp:98] startAP(): [APSettingsService] Starting captive portal on 192.168.4.1
```
I wanted to let know that the AP connectivity is a bit dodgy. I can't get the I2C communication for any number of reasons but I think I was suppose to put the 4.7k resistor inline for the SDA and SCL connections.

runeharlyk, 2024-06-19:
I will look into the connectivity. For the I2C, there should not be anything between SCL and SDA connections.

Ian-cmd-ops, 2024-06-20:
[image]

> I will look into the connectivity. For the I2C, there should not be anything between SCL and SDA connections.

https://a.co/d/037ObbSe
I'm using this serial wombat as an I2C bus. I can't get read any serial data off the board when it's connected. do you have diagram to read the info from the board if I have a USB to TTL adapter.

runeharlyk, 2024-06-20:
The Serial Wombat you linked to is not a device in of it self, the description mentions *Board contains no electronics*. Its only made to make it simpler to hook up other devices - like a breadboard. You have to connect the board to the esp32, and then the I2C devices (MPU6050, PCA9685 etc) to the "Serial Wombat".

Ian-cmd-ops, 2024-06-20:
For clarification, I have connected that i2c and connected the pins in the same order as the silk screen. I'm unable to verify i2c data when it was connected. I had no comms with IMU when connected to the AP with the IMU connected to the Serial Wombat. Additional I couldn't verify comms with PCA9685. I thought it was my wiring was the issue but I triple checked it. I've exhausted most possible troubleshooting issues from my end. I'll attempt to see it the issue continues on my breadboard. Cheers mate.

runeharlyk, 2024-07-18:
Does the issue persist or can i close it?

closed 2024-07-30 by runeharlyk