runeharlyk/SpotMicroESP32-Leika #73: cannot imort icon

Mark11200, 2024-07-19:
6:03:37 PM [vite] Pre-transform error: Failed to resolve import "~icons/tabler/circle-x" from "src/lib/components/toasts/Toast.svelte". Does the file exist?
6:03:37 PM [vite] Pre-transform error: Failed to resolve import "~icons/tabler/wifi-off" from "src/routes/statusbar.svelte". Does the file exist?
6:03:38 PM [vite] Pre-transform error: Failed to resolve import "~icons/mdi/github" from "src/routes/menu.svelte". Does the file exist?

🌼   daisyUI 4.12.10
├─ ✔︎ 2 themes added            https://daisyui.com/docs/themes
╰─ ❤︎ Support daisyUI project:  https://opencollective.com/daisyui

6:03:40 PM [vite] Internal server error: Failed to resolve import "~icons/tabler/login" from "src/routes/login.svelte". Does the file exist?
  Plugin: vite:import-analysis
  File: C:/Users/mark/Desktop/app/src/routes/login.svelte:7:19
  43 |  import { notifications } from "$lib/components/toasts/notifications";
  44 |  import { fade, fly } from "svelte/transition";
  45 |  import Login from "~icons/tabler/login";
     |                     ^
  46 |  import { api } from "$lib/api";
  47 |  const file = "C:/Users/mark/Desktop/app/src/routes/login.svelte";
      at TransformPluginContext._formatError (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:49748:41)
      at TransformPluginContext.error (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:49743:16)
      at normalizeUrl (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:64333:23)
      at async file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:64465:39
      at async Promise.all (index 7)
      at async TransformPluginContext.transform (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:64392:7)
      at async PluginContainer.transform (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:49589:18)
      at async loadAndTransform (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:52411:27)
6:03:40 PM [vite] Pre-transform error: Failed to resolve import "~icons/tabler/login" from "src/routes/login.svelte". Does the file exist?
6:03:41 PM [vite] Pre-transform error: Failed to resolve import "~icons/tabler/x" from "src/lib/components/ConfirmDialog.svelte". Does the file exist?
6:03:41 PM [vite] Internal server error: Failed to resolve import "~icons/tabler/circle-x" from "src/lib/components/toasts/Toast.svelte". Does the file exist?
  Plugin: vite:import-analysis
  File: C:/Users/mark/Desktop/app/src/lib/components/toasts/Toast.svelte:5:19
  47 |  import { fly } from 'svelte/transition';
  48 |  import { notifications } from '$lib/components/toasts/notifications';
  49 |  import error from '~icons/tabler/circle-x';
     |                     ^
  50 |  import success from '~icons/tabler/circle-check';
  51 |  import warning from '~icons/tabler/alert-triangle';
      at TransformPluginContext._formatError (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:49748:41)
      at TransformPluginContext.error (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:49743:16)
      at normalizeUrl (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:64333:23)
      at async file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:64465:39
      at async Promise.all (index 5)
      at async TransformPluginContext.transform (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:64392:7)
      at async PluginContainer.transform (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:49589:18)
      at async loadAndTransform (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:52411:27)
      at async viteTransformMiddleware (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:62175:24)
6:03:42 PM [vite] Pre-transform error: Failed to resolve import "~icons/tabler/wifi" from "src/lib/components/RSSIIndicator.svelte". Does the file exist?
6:03:42 PM [vite] Pre-transform error: Failed to resolve import "~icons/tabler/battery" from "src/lib/components/BatteryIndicator.svelte". Does the file exist?
6:03:42 PM [vite] Pre-transform error: Failed to resolve import "~icons/tabler/refresh-alert" from "src/lib/components/UpdateIndicator.svelte". Does the file exist?
6:03:43 PM [vite] Internal server error: Failed to resolve import "~icons/mdi/github" from "src/routes/menu.svelte". Does the file exist?
  Plugin: vite:import-analysis
  File: C:/Users/mark/Desktop/app/src/routes/menu.svelte:3:23
  44 |  import "svelte/internal/disclose-version";
  45 |  import logo from "$lib/assets/logo512.png";
  46 |  import MdiGithub from "~icons/mdi/github";
     |                         ^
  47 |  import MdiConnection from "~icons/mdi/connection";
  48 |  import Users from "~icons/mdi/users";
      at TransformPluginContext._formatError (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:49748:41)
      at TransformPluginContext.error (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:49743:16)
      at normalizeUrl (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:64333:23)
      at process.processTicksAndRejections (node:internal/process/task_queues:95:5)    
      at async file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:64465:39
      at async Promise.all (index 3)
      at async TransformPluginContext.transform (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:64392:7)
      at async PluginContainer.transform (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:49589:18)
      at async loadAndTransform (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:52411:27)
      at async viteTransformMiddleware (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:62175:24)
6:03:43 PM [vite] Internal server error: Failed to resolve import "~icons/tabler/wifi-off" from "src/routes/statusbar.svelte". Does the file exist?
  Plugin: vite:import-analysis
  File: C:/Users/mark/Desktop/app/src/routes/statusbar.svelte:6:21
[... 251 lines cut ...]
9:36:50 PM [vite] Internal server error: Failed to resolve import "~icons/tabler/login" from "src/routes/login.svelte". Does the file exist?
  Plugin: vite:import-analysis
  File: C:/Users/mark/Desktop/app/src/routes/login.svelte:7:19
  43 |  import { notifications } from "$lib/components/toasts/notifications";
  44 |  import { fade, fly } from "svelte/transition";
  45 |  import Login from "~icons/tabler/login";
     |                     ^
  46 |  import { api } from "$lib/api";
  47 |  const file = "C:/Users/mark/Desktop/app/src/routes/login.svelte";
      at TransformPluginContext._formatError (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:49748:41)
      at TransformPluginContext.error (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:49743:16)
      at normalizeUrl (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:64333:23)
      at process.processTicksAndRejections (node:internal/process/task_queues:95:5)    
      at async file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:64465:39
      at async Promise.all (index 7)
      at async TransformPluginContext.transform (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:64392:7)
      at async PluginContainer.transform (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:49589:18)
      at async loadAndTransform (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:52411:27)
      at async viteTransformMiddleware (file:///C:/Users/mark/Desktop/app/node_modules/vite/dist/node/chunks/dep-D8YhmIY-.js:62175:24)
9:36:50 PM [vite] Pre-transform error: Failed to resolve import "~icons/tabler/x" from "src/lib/components/ConfirmDialog.svelte". Does the file exist?
[... cut ...]
9:36:50 PM [vite] Pre-transform error: Failed to resolve import "~icons/tabler/refresh-alert" from "src/lib/components/UpdateIndicator.svelte". Does the file exist?
9:36:50 PM [vite] Pre-transform error: Failed to resolve import "~icons/tabler/battery" from "src/lib/components/BatteryIndicator.svelte". Does the file exist?
9:36:51 PM [vite] Pre-transform error: Failed to resolve import "~icons/tabler/x" from "src/lib/components/ConfirmDialog.svelte". Does the file exist?
9:36:51 PM [vite] Pre-transform error: Failed to resolve import "~icons/tabler/x" from "src/lib/components/GithubUpdateDialog.svelte". Does the file exist?

runeharlyk, 2024-07-20:
Hi. I haven't found a good way to handle those includes. To save space on the esp32, I have compressed the model files to a single zip file. This means that when a client opens the app for the first time, the zip fill will be unpacked and stored in localstorage. The following times a function will intercept the fetch of the models files and return them from localstorage instead of throwing 404 errors.

Mark11200, 2024-07-21:
thanks ..i just rebuild it again the missing stl is ok now but in the preview i get connection issue it always get disconnected and i cannot configure any of the settings the only one  i known working is the camera when clicking i get the preview but in the control i get nothing even the visual robot did not move when moving the joystick or the heigth. ill try the vite.config.ts changing the ip. thanks  again for your help.

Mark11200, 2024-07-21:
can u provide me example of what should i pun on the vite.config.ts i dony know what should i put. by the way i solve the stl not found by intalling unzip .

runeharlyk, 2024-07-21:
The reason the model did not move, is because It need a connection to the device. To test the kinematics before connecting the robot, you can make it use the internal kinematics. This option is found under the ThreeJS GUI 'Controls'

You can change `spot-micro.local` with the IP address of the device in `vite.config.ts`. 

Have you run `pnpm install` or `npm install` from /app? That should install all depencies.

Mark11200, 2024-07-21:
ok thanks i just run  the npm install. then the  npm run dev  to open the apps.

Mark11200, 2024-07-21:
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    33][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   482][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   505][I][ESP32SvelteKit.cpp:74] begin(): [Running Firmware Version: %s] 0.0.1
[   524][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   590][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
[   643][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 3 - STA_STOP
[   757][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: WiFiSettings
[   802][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: angles
[   802][I][CameraService.cpp:80] InitializeCamera(): Initializing camera
[  1211][I][CameraService.cpp:83] InitializeCamera(): [CameraService] Camera probe successful
[  1215][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: CameraSettings
[  1219][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: peripheralSettings
[  1472][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: i2cScan
[  1473][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=14 scl=15 freq=100000
[  1479][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  2499][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  3512][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  4525][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  5529][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  6533][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  7537][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  8541][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  9544][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 10649][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 11767][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 11767][E][Peripherals.h:144] begin(): [IMUService] MPU initialize failed
[ 12772][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 13776][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 14780][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 15784][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 16790][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 16791][E][Peripherals.h:160] begin(): [IMUService] BMP initialize failed
[ 16792][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[ 16799][E][Peripherals.h:166] begin(): [Peripherals] ADS1015/ADS1115 not found
[ 16808][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[ 17827][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 17837][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: servoConfiguration
[ 19013][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 20026][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 20033][I][ServoController.h:154] configure(): [ServoController] Configured with oscillator frequency 27000000 and PWM frequency 50
[ 20034][I][APSettingsService.cpp:69] startAP(): [APSettingsService] Starting software access point
[ 20063][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 20073][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[ 20785][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 11 - AP_STOP
[ 20785][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[ 20788][I][APSettingsService.cpp:77] startAP(): [APSettingsService] Starting captive portal on 192.168.4.1
[ 22158][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 23162][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 24165][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 25168][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 26171][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 27174][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 28177][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
---- Closed the serial port COM3 ----
this is i got in the serial monitor

Mark11200, 2024-07-21:
Processing esp32-camera (board: esp32cam; platform: espressif32 @ 6.6.0; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
running: build_app.py
Newest file: 2024-07-19 01:34:12.411131, output file: 2024-07-19 01:34:21.249519
SSL Cert Store: Certificate bundle downloaded to: ssl_certs\cacert.pem
SSL Cert Store: Parsing certificates from ssl_certs\cacert.pem
SSL Cert Store: Successfully added 48 certificates
SSL Cert Store: Successfully added 48 certificates in total
SSL Cert Store: Successfully created src\certs\x509_crt_bundle.bin
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 (6.6.0) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 54 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 7.1.0
|-- NewPing @ 1.9.7
|-- I2Cdevlib-MPU6050 @ 1.0.0
|-- Adafruit SSD1306 @ 2.5.10
|-- Adafruit GFX Library @ 1.11.9
|-- Adafruit BusIO @ 1.16.1
|-- Adafruit PWM Servo Driver Library @ 2.4.1
|-- Adafruit ST7735 and ST7789 Library @ 1.10.4
|-- FastLED @ 3.7.0
|-- Adafruit ADS1X15 @ 2.5.0
|-- SPI @ 2.0.0
|-- ESP32-sveltekit
|-- PsychicHttp @ 1.0.1
Building in release mode
Building FS image from 'data' directory to .pio\build\esp32-camera\littlefs.bin
============================================= [SUCCESS] Took 13.34 seconds =============================================

Environment    Status    Duration
-------------  --------  ------------
esp32-camera   SUCCESS   00:00:13.340
============================================== 1 succeeded in 00:00:13.340 ==============================================
 *  Terminal will be reused by tasks, press any key to close it. 
heres the build filesystem image  then i upload it using the filesystem upload

runeharlyk, 2024-07-21:
The next step is to connect the device to your internet. The device will host an access point you can connect to and configure the network settings.

Have you selected the feature you would like to have enabled in `factory_settings.h`

Mark11200, 2024-07-21:
i dont change any of the setting in th factory_settings.h i dont know what should i change

Mark11200, 2024-07-21:
in the vite.config.ts what ip should i put the 192.168.4.1 or any number

runeharlyk, 2024-07-21:
I'm sorry I meant in `features.h`, by default it's assumed a lot of sensors are enabled and connected. You can disable those.

You have to connect the the access point hosted by the ESP32 to configure the network settings. When those settings are applied and the device have connected to your network, its IP address will be printed to the monitor. That's the IP address to replace `spot-micro.local` with

Mark11200, 2024-07-21:
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    41][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   490][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   514][I][ESP32SvelteKit.cpp:74] begin(): [Running Firmware Version: %s] 0.0.1
[   531][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   597][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
[   630][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 3 - STA_STOP
[   730][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: WiFiSettings
[   758][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: angles
[   758][I][CameraService.cpp:80] InitializeCamera(): Initializing camera
[  1166][I][CameraService.cpp:83] InitializeCamera(): [CameraService] Camera probe successful
[  1171][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: CameraSettings
[  1175][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: peripheralSettings
[  1294][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: i2cScan
[  1294][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=14 scl=15 freq=100000
[  1300][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  1341][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  1366][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: servoConfiguration
[  1498][I][ServoController.h:154] configure(): [ServoController] Configured with oscillator frequency 27000000 and PWM frequency 50
[ 10030][I][APSettingsService.cpp:69] startAP(): [APSettingsService] Starting software access point
[ 10051][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 10061][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[ 10768][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 11 - AP_STOP
[ 10768][I][APSettingsService.cpp:77] startAP(): [APSettingsService] Starting captive portal on 192.168.4.1
[ 10768][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[ 58103][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
[ 58863][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
[ 59933][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 52
[ 60036][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 52
[ 62140][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 52
[ 62434][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 52
[ 64589][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 52
[ 64725][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 53
[ 64731][I][EventSocket.cpp:60] onWSOpen(): [EventSocket] ws[192.168.4.2][53] connect
[ 80209][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 54
[ 90169][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
[ 90238][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 55
[120335][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
[120404][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 55
[150464][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
[150530][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 55
[180613][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
[180677][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 55
[210757][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
[210826][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 55
[240939][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
[241010][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 55
[271123][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
[271187][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 55
[301298][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
[301363][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 55
[332334][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
[332536][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 55
[362844][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
[362906][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 55

heres i get in the  serial monitor  i type 0 on the other sensor only the servo has 1 in the  features.ini to test only the servo but i did not get any response

Mark11200, 2024-07-22:
PS C:\Users\mark\Desktop\new spot\SpotMicroESP32-Leika-master\SpotMicroESP32-Leika\app> npm run dev -- --open

> spot_micro_controller@0.0.1 dev
> vite dev --open

5:08:57 PM [vite-plugin-svelte] WARNING: The following packages have a svelte field in their package.json but no exports condition for svelte.

svelte-dnd-list@0.1.8
svelte-focus-trap@1.2.0

Please see https://github.com/sveltejs/vite-plugin-svelte/blob/main/docs/faq.md#missing-exports-condition for details.   

  VITE v5.3.4  ready in 4832 ms

  ➜  Local:   http://localhost:5173/
  ➜  Network: use --host to expose
  ➜  press h + enter to show help
o

🌼   daisyUI 4.12.10
├─ ✔︎ 2 themes added            https://daisyui.com/docs/themes
╰─ ❤︎ Support daisyUI project:  https://opencollective.com/daisyui

5:09:32 PM [vite] http proxy error: /api/features
Error: getaddrinfo ENOTFOUND spot-micro.local
    at GetAddrInfoReqWrap.onlookupall [as oncomplete] (node:dns:120:26)
    at GetAddrInfoReqWrap.callbackTrampoline (node:internal/async_hooks:130:17)
5:09:45 PM [vite] http proxy error: /api/features
Error: getaddrinfo ENOTFOUND spot-micro.local
    at GetAddrInfoReqWrap.onlookupall [as oncomplete] (node:dns:120:26)
    at GetAddrInfoReqWrap.callbackTrampoline (node:internal/async_hooks:130:17) (x2)
5:09:59 PM [vite] http proxy error: /api/features
Error: getaddrinfo ENOTFOUND spot-micro.local
    at GetAddrInfoReqWrap.onlookupall [as oncomplete] (node:dns:120:26)
    at GetAddrInfoReqWrap.callbackTrampoline (node:internal/async_hooks:130:17) (x3)

Mark11200, 2024-07-23:
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[     9][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   458][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   471][I][ESP32SvelteKit.cpp:74] begin(): [Running Firmware Version: %s] 0.0.1
[   481][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   546][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
[   560][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 3 - STA_STOP
[   609][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: WiFiSettings
[   629][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: angles
[   629][I][CameraService.cpp:80] InitializeCamera(): Initializing camera
[  1038][I][CameraService.cpp:83] InitializeCamera(): [CameraService] Camera probe successful
[  1042][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: CameraSettings
[  1046][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: peripheralSettings
[  1128][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: i2cScan
[  1128][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=14 scl=15 freq=100000
[  1134][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  3154][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  3178][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: servoConfiguration
[  3301][I][ServoController.h:154] configure(): [ServoController] Configured with oscillator frequency 27000000 and PWM frequency 50
[ 10029][I][APSettingsService.cpp:69] startAP(): [APSettingsService] Starting software access point
[ 10044][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 10051][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[ 10754][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 11 - AP_STOP
[ 10754][I][APSettingsService.cpp:77] startAP(): [APSettingsService] Starting captive portal on 192.168.4.1
[ 10754][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[ 21770][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
[ 21991][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
[ 22679][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 52
[ 22707][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 53
[ 22734][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 54
[ 22744][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 53
[ 23015][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 54
[ 23139][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 53
E (23194) httpd_txrx: httpd_resp_send_err: error calling setsockopt : 22
[ 23202][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 53
[ 23769][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 53
[ 23785][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 53
[ 23795][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 53
[ 23804][I][EventSocket.cpp:60] onWSOpen(): [EventSocket] ws[192.168.4.2][53] connect
[ 25631][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 54
[ 25663][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 54
[ 25824][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 54
[ 25882][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 54
[ 26731][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 54
[ 29920][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
[ 29941][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 55
[ 37977][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
[ 37996][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 55
[ 54038][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
[ 54057][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 55
[ 83970][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
[ 84031][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 55
[ 84115][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
E (84274) httpd_txrx: httpd_resp_send_err: error calling setsockopt : 22
[ 84282][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 55
[ 84300][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
E (84536) httpd_txrx: httpd_resp_send_err: error calling setsockopt : 22
[ 84543][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 55

---- Closed the serial port COM3 ----

runeharlyk, 2024-07-24:
Have you connected the device to your network and updated the proxy location in the vite config with the assigned ip address?

Mark11200, 2024-07-24:
yes i conected it to my wifi network. but i dont know wat should proxy i will put on vitr. config. heres what i got in serial

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    37][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   485][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   499][I][ESP32SvelteKit.cpp:74] begin(): [Running Firmware Version: %s] 0.0.1
[   510][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   576][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
[   614][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 3 - STA_STOP
[   680][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: WiFiSettings
[   716][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: angles
[   717][I][CameraService.cpp:80] InitializeCamera(): Initializing camera
[  1127][I][CameraService.cpp:83] InitializeCamera(): [CameraService] Camera probe successful
[  1131][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: CameraSettings
[  1135][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: peripheralSettings
[  1355][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: i2cScan
[  1356][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=14 scl=15 freq=100000
[  1361][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  3381][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  3405][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: servoConfiguration
[  3572][I][ServoController.h:154] configure(): [ServoController] Configured with oscillator frequency 27000000 and PWM frequency 50
[ 10017][I][APSettingsService.cpp:69] startAP(): [APSettingsService] Starting software access point
[ 10031][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 10038][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[ 10738][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 11 - AP_STOP
[ 10738][I][APSettingsService.cpp:77] startAP(): [APSettingsService] Starting captive portal on 192.168.4.1
[ 10739][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[ 30212][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
[ 33050][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 33052][I][WiFiSettingsService.cpp:94] connectToWiFi(): [WiFiSettingsService] 1 networks found.
[ 33054][I][WiFiSettingsService.cpp:134] connectToWiFi(): [WiFiSettingsService] Connecting to strongest network: TP-Link_5CA6
[ 33840][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 33840][I][WiFiStatus.cpp:33] onStationModeConnected(): [WiFiStatus] WiFi Connected.
[ 34458][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 34458][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 192.168.0.102, MASK: 255.255.255.0, GW: 192.168.0.1
[ 34464][I][WiFiStatus.cpp:42] onStationModeGotIP(): [WiFiStatus] WiFi Got IP. localIP=192.168.0.102, hostName=spot-micro
[ 36555][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 52
[ 36561][I][EventSocket.cpp:60] onWSOpen(): [EventSocket] ws[192.168.0.100][52] connect
[ 36562][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 53
[ 36573][I][EventSocket.cpp:60] onWSOpen(): [EventSocket] ws[192.168.0.100][53] connect
[ 36576][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 52
[ 36583][I][EventSocket.cpp:67] onWSClose(): [EventSocket] ws[192.168.0.100][52] disconnect
[ 37094][I][Peripherals.h:208] scanI2C(): [Peripherals] I2C device found at address 0x40
[ 37101][I][Peripherals.h:208] scanI2C(): [Peripherals] I2C device found at address 0x68
[ 37105][I][Peripherals.h:215] scanI2C(): [Peripherals] Scan complete - Found 2 devices
[ 37107][I][Peripherals.h:198] emitI2C(): [Peripherals] Emitting I2C scan results, 53 1
---- Closed the serial port COM3 ----

runeharlyk, 2024-07-24:
Perfect! The device ip address is `192.168.0.102`. This value should replace `spot-micro.local` in `vite.config.ts`.

Like this:
```ts
...
    server: {
        proxy: {
            '/api': {
                target: 'http://192.168.0.102/', // here
                changeOrigin: true,
                ws: true
            },
            '/ws': {
                target: 'ws://192.168.0.102/', // here
                changeOrigin: true,
                ws: true
            }
        }
  ...
```

Mark11200, 2024-07-24:
Thanks for the help i'll try to edit the vite.config.ts

Mark11200, 2024-07-24:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    37][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   485][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   499][I][ESP32SvelteKit.cpp:74] begin(): [Running Firmware Version: %s] 0.0.1
[   510][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   576][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
[   612][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 3 - STA_STOP
[   672][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: WiFiSettings
[   699][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: angles
[   699][I][CameraService.cpp:80] InitializeCamera(): Initializing camera
[  1110][I][CameraService.cpp:83] InitializeCamera(): [CameraService] Camera probe successful
[  1114][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: CameraSettings
[  1118][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: peripheralSettings
[  1230][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: i2cScan
[  1231][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=14 scl=15 freq=100000
[  1236][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  3256][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  3280][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: servoConfiguration
[  3404][I][ServoController.h:154] configure(): [ServoController] Configured with oscillator frequency 27000000 and PWM frequency 50
[ 10017][I][APSettingsService.cpp:69] startAP(): [APSettingsService] Starting software access point
[ 10029][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 10036][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[ 10728][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 11 - AP_STOP
[ 10729][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[ 10728][I][APSettingsService.cpp:77] startAP(): [APSettingsService] Starting captive portal on 192.168.4.1
[ 30005][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
[ 32842][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 32844][I][WiFiSettingsService.cpp:94] connectToWiFi(): [WiFiSettingsService] 1 networks found.
[ 32848][I][WiFiSettingsService.cpp:134] connectToWiFi(): [WiFiSettingsService] Connecting to strongest network: TP-Link_5CA6
[ 33638][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 33639][I][WiFiStatus.cpp:33] onStationModeConnected(): [WiFiStatus] WiFi Connected.
[ 34376][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 34377][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 192.168.0.102, MASK: 255.255.255.0, GW: 192.168.0.1
[ 34382][I][WiFiStatus.cpp:42] onStationModeGotIP(): [WiFiStatus] WiFi Got IP. localIP=192.168.0.102, hostName=spot-micro
[ 35133][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 52
[ 36111][I][EventSocket.cpp:60] onWSOpen(): [EventSocket] ws[192.168.0.100][52] connect
[ 36315][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 53
[ 36627][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 54
[ 36692][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 53
[ 36701][I][EventSocket.cpp:60] onWSOpen(): [EventSocket] ws[192.168.0.100][54] connect
[ 37069][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 53
[ 37109][I][EventSocket.cpp:60] onWSOpen(): [EventSocket] ws[192.168.0.100][53] connect
[ 37648][I][Peripherals.h:208] scanI2C(): [Peripherals] I2C device found at address 0x40
[ 37656][I][Peripherals.h:208] scanI2C(): [Peripherals] I2C device found at address 0x68
[ 37661][I][Peripherals.h:215] scanI2C(): [Peripherals] Scan complete - Found 2 devices
[ 37661][I][Peripherals.h:198] emitI2C(): [Peripherals] Emitting I2C scan results, 53 1
[ 37675][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
[ 37682][I][EventSocket.cpp:60] onWSOpen(): [EventSocket] ws[192.168.0.100][55] connect
[ 38105][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 56
[ 38250][I][EventSocket.cpp:60] onWSOpen(): [EventSocket] ws[192.168.0.100][56] connect
[ 38357][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 57
[ 38371][I][EventSocket.cpp:60] onWSOpen(): [EventSocket] ws[192.168.0.100][57] connect
[ 38724][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 58
[ 38945][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 58
[ 39484][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 58
[ 39490][I][EventSocket.cpp:60] onWSOpen(): [EventSocket] ws[192.168.0.100][58] connect
[ 42913][I][APSettingsService.cpp:85] stopAP(): [APSettingsService] Stopping captive portal
[ 42915][I][APSettingsService.cpp:90] stopAP(): [APSettingsService] Stopping AP
[ 42919][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 11 - AP_STOP
[... cut ...]
[118660][I][EventSocket.cpp:67] onWSClose(): [EventSocket] ws[192.168.0.100][53] disconnect
[119651][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 53
[119658][I][EventSocket.cpp:60] onWSOpen(): [EventSocket] ws[192.168.0.100][53] connect
[120625][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 55
[120631][I][EventSocket.cpp:60] onWSOpen(): [EventSocket] ws[192.168.0.100][55] connect
---- Closed the serial port COM3 ----

heres what i got

Mark11200, 2024-07-24:
Backtrace: 0xfffffffe:0x800924ac |<-CORRUPTED

ELF file SHA256: 898b01d3ebc47ce5

Mark11200, 2024-07-24:
2:19:19 AM [vite] ws proxy socket error:
Error: write ECONNABORTED
    at afterWriteDispatched (node:internal/stream_base_commons:161:15)
    at writeGeneric (node:internal/stream_base_commons:152:3)
    at Socket._writeGeneric (node:net:953:11)
    at Socket._write (node:net:965:8)
    at writeOrBuffer (node:internal/streams/writable:564:12)
    at _write (node:internal/streams/writable:493:10)
    at Writable.write (node:internal/streams/writable:502:10)
    at Socket.ondata (node:internal/streams/readable:1007:22)
    at Socket.emit (node:events:519:28)
    at addChunk (node:internal/streams/readable:559:12)
2:20:32 AM [vite] ws proxy error:
Error: read ECONNRESET
    at TCP.onStreamRead (node:internal/stream_base_commons:218:20)
    at TCP.callbackTrampoline (node:internal/async_hooks:130:17)
2:20:32 AM [vite] ws proxy error:
Error: read ECONNRESET
    at TCP.onStreamRead (node:internal/stream_base_commons:218:20)
    at TCP.callbackTrampoline (node:internal/async_hooks:130:17) (x2)
2:20:32 AM [vite] ws proxy error:
Error: read ECONNRESET
    at TCP.onStreamRead (node:internal/stream_base_commons:218:20)
    at TCP.callbackTrampoline (node:internal/async_hooks:130:17) (x3)
2:20:40 AM [vite] ws proxy error:
Error: connect ETIMEDOUT 192.168.0.102:80
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1606:16)
    at TCPConnectWrap.callbackTrampoline (node:internal/async_hooks:130:17)
2:21:13 AM [vite] ws proxy socket error:
Error: write ECONNABORTED
    at afterWriteDispatched (node:internal/stream_base_commons:161:15)
    at writeGeneric (node:internal/stream_base_commons:152:3)
    at Socket._writeGeneric (node:net:953:11)
    at Socket._write (node:net:965:8)
    at writeOrBuffer (node:internal/streams/writable:564:12)
    at _write (node:internal/streams/writable:493:10)
    at Writable.write (node:internal/streams/writable:502:10)
    at Socket.ondata (node:internal/streams/readable:1007:22)
    at Socket.emit (node:events:519:28)
    at addChunk (node:internal/streams/readable:559:12)

runeharlyk, 2024-07-24:
See if it helps restarting the server. After changing the proxy address the server usually needs to be restarted.

Mark11200, 2024-07-25:
🌼   daisyUI 4.12.10
├─ ✔︎ 2 themes added            https://daisyui.com/docs/themes
╰─ ❤︎ Support daisyUI project:  https://opencollective.com/daisyui

10:11:08 AM [vite] ws proxy error:
Error: read ECONNRESET
    at TCP.onStreamRead (node:internal/stream_base_commons:218:20)
    at TCP.callbackTrampoline (node:internal/async_hooks:130:17)
10:11:15 AM [vite] ws proxy error:
Error: connect ETIMEDOUT 192.168.0.102:80
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1606:16)
    at TCPConnectWrap.callbackTrampoline (node:internal/async_hooks:130:17)

Mark11200, 2024-07-25:
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[     9][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   458][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   472][I][ESP32SvelteKit.cpp:74] begin(): [Running Firmware Version: %s] 0.0.1
[   485][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   550][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
[   556][E][vfs_api.cpp:105] open(): /littlefs/config/wifiSettings.json does not exist, no permits for creation
[   584][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 3 - STA_STOP
[   618][E][vfs_api.cpp:105] open(): /littlefs/config/apSettings.json does not exist, no permits for creation
[   650][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: WiFiSettings
[   657][E][vfs_api.cpp:105] open(): /littlefs/config/ntpSettings.json does not exist, no permits for creation
[   750][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: angles
[   750][I][CameraService.cpp:80] InitializeCamera(): Initializing camera
[  1161][I][CameraService.cpp:83] InitializeCamera(): [CameraService] Camera probe successful
[  1165][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: CameraSettings
[  1169][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: peripheralSettings
[  1191][E][vfs_api.cpp:105] open(): /littlefs/config/peripheral.json does not exist, no permits for creation
[  1279][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: i2cScan
[  1280][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=14 scl=15 freq=100000
[  1285][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  3305][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  3329][D][EventSocket.cpp:178] onSubscribe(): [EventSocket] onSubscribe for event: servoConfiguration
[  3346][E][vfs_api.cpp:105] open(): /littlefs/config/servoConfig.json does not exist, no permits for creation
[  3346][I][ServoController.h:78] update(): [ControllerSettings] No servos found, adding default servos
[  3354][I][ServoController.h:80] update(): [ControllerSettings] Adding servo 0
[  3362][I][ServoController.h:80] update(): [ControllerSettings] Adding servo 1
[  3368][I][ServoController.h:80] update(): [ControllerSettings] Adding servo 2
[  3376][I][ServoController.h:80] update(): [ControllerSettings] Adding servo 3
[  3382][I][ServoController.h:80] update(): [ControllerSettings] Adding servo 4
[  3390][I][ServoController.h:80] update(): [ControllerSettings] Adding servo 5
[  3396][I][ServoController.h:80] update(): [ControllerSettings] Adding servo 6
[  3404][I][ServoController.h:80] update(): [ControllerSettings] Adding servo 7
[  3410][I][ServoController.h:80] update(): [ControllerSettings] Adding servo 8
[  3418][I][ServoController.h:80] update(): [ControllerSettings] Adding servo 9
[  3424][I][ServoController.h:80] update(): [ControllerSettings] Adding servo 10
[  3432][I][ServoController.h:80] update(): [ControllerSettings] Adding servo 11
[  3620][I][ServoController.h:154] configure(): [ServoController] Configured with oscillator frequency 27000000 and PWM frequency 50
[ 10018][I][APSettingsService.cpp:69] startAP(): [APSettingsService] Starting software access point
[ 10031][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 10038][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[ 10723][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 11 - AP_STOP
[ 10724][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[ 10724][I][APSettingsService.cpp:77] startAP(): [APSettingsService] Starting captive portal on 192.168.4.1
[ 30009][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
[ 32846][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 32847][I][WiFiSettingsService.cpp:94] connectToWiFi(): [WiFiSettingsService] 2 networks found.
[ 32851][I][WiFiSettingsService.cpp:134] connectToWiFi(): [WiFiSettingsService] Connecting to strongest network: TP-Link_5CA6
[ 33686][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 33686][I][WiFiStatus.cpp:33] onStationModeConnected(): [WiFiStatus] WiFi Connected.
[ 34364][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 34365][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 192.168.0.102, MASK: 255.255.255.0, GW: 192.168.0.1
[ 34370][I][WiFiStatus.cpp:42] onStationModeGotIP(): [WiFiStatus] WiFi Got IP. localIP=192.168.0.102, hostName=spot-micro
[ 34566][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 52
[ 34621][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 52
[ 34631][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 52
[ 34697][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 52
[ 34938][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 52
[ 34943][I][EventSocket.cpp:60] onWSOpen(): [EventSocket] ws[192.168.0.100][52] connect
[... cut ...]
[153761][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 53
[157671][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 53
[157874][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 53
[162571][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 53
[162796][I][PsychicHttpServer.cpp:261] closeCallback(): [psychic] Client disconnected 53
[167690][I][PsychicHttpServer.cpp:235] openCallback(): [psychic] New client connected 53
[167894][I][PsychicHttpServer.cpp:261] closeCallback(

Mark11200, 2024-07-25:
The servo on works only in peripherals. 

in I2C it detects the pca9685 and the mpu 6050 the rest i disable it 
[40] PCA9685 - 16-channel PWM driver default address
[68] MPU6050 - Six-Axis (Gyro + Accelerometer) MEMS MotionTracking™ Devices

when i open the portal ni my android phone by typing the http://192.168.0.102/ or the http://192.168.4.1/ i get the That URI does not exist. so i can only see it in my laptop i cant get the portal in my android  for testing

runeharlyk, 2024-07-25:
Currently the servos can only be configured via direct control from peripherals. I'm working a calibration guide and connecting the motion service with the servo controller. This is done on the branch `servo-motion`. 

Are you positive your phone is on the same network as your phone?

Mark11200, 2024-07-25:
yes im in the same network even on my laptop i cant access it gives me the same url not exist. i can acess it in my laptop by the http://localhost:5173/  when i type the npm run dev command

Mark11200, 2024-07-26:
Processing esp32dev (board: esp32dev; platform: espressif32 @ 6.6.0; framework: arduino)
-----------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
running: build_app.py
Newest file: 2024-07-24 23:19:57.028848, output file: 2024-07-24 23:22:39.882852
SSL Cert Store: Certificate bundle downloaded to: ssl_certs\cacert.pem
SSL Cert Store: Parsing certificates from ssl_certs\cacert.pem
SSL Cert Store: Successfully added 48 certificates
SSL Cert Store: Successfully added 48 certificates in total
SSL Cert Store: Successfully created src\certs\x509_crt_bundle.bin
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.6.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 53 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 7.1.0
|-- NewPing @ 1.9.7
|-- I2Cdevlib-MPU6050 @ 1.0.0
|-- Adafruit SSD1306 @ 2.5.10
|-- Adafruit GFX Library @ 1.11.10
|-- Adafruit BusIO @ 1.16.1
|-- Adafruit PWM Servo Driver Library @ 2.4.1
|-- Adafruit ST7735 and ST7789 Library @ 1.10.4
|-- FastLED @ 3.7.0
|-- Adafruit ADS1X15 @ 2.5.0
|-- SPI @ 2.0.0
|-- ESP32-sveltekit
|-- PsychicHttp @ 1.0.1
Building in release mode
Converting .pio\build\esp32dev\x509_crt_bundle.bin.txt.o
Compiling .pio\build\esp32dev\lib304\FastLED\FastLED.cpp.o
Compiling .pio\build\esp32dev\lib304\FastLED\bitswap.cpp.o
Compiling .pio\build\esp32dev\lib304\FastLED\colorpalettes.cpp.o
Compiling .pio\build\esp32dev\lib304\FastLED\colorutils.cpp.o
Compiling .pio\build\esp32dev\lib304\FastLED\five_bit_hd_gamma.cpp.o
Compiling .pio\build\esp32dev\lib304\FastLED\hsv2rgb.cpp.o
Compiling .pio\build\esp32dev\lib304\FastLED\lib8tion.cpp.o
Compiling .pio\build\esp32dev\lib304\FastLED\noise.cpp.o
In file included from .pio/libdeps/esp32dev/FastLED/src/FastLED.h:79,
                 from .pio/libdeps/esp32dev/FastLED/src/five_bit_hd_gamma.cpp:1:
.pio/libdeps/esp32dev/FastLED/src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~Compiling .pio\build\esp32dev\lib304\FastLED\platforms.cpp.o
Compiling .pio\build\esp32dev\lib304\FastLED\platforms\esp\32\clockless_rmt_esp32.cpp.o  
Compiling .pio\build\esp32dev\lib304\FastLED\power_mgt.cpp.o
~~~~~~~~~~~~~
Compiling .pio\build\esp32dev\lib304\FastLED\wiring.cpp.o
Compiling .pio\build\esp32dev\src\main.cpp.o
Building .pio\build\esp32dev\bootloader.bin
Generating partitions .pio\build\esp32dev\partitions.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\esp32dev\liba06\NewPing\NewPing.cpp.o
Compiling .pio\build\esp32dev\lib1cc\Wire\Wire.cpp.o
Compiling .pio\build\esp32dev\lib35e\I2Cdevlib-Core\I2Cdev.cpp.o
Archiving .pio\build\esp32dev\liba06\libNewPing.a
Archiving .pio\build\esp32dev\lib1cc\libWire.a
Compiling .pio\build\esp32dev\lib382\I2Cdevlib-MPU6050\MPU6050.cpp.o
Archiving .pio\build\esp32dev\lib35e\libI2Cdevlib-Core.a
Compiling .pio\build\esp32dev\lib382\I2Cdevlib-MPU6050\MPU6050_6Axis_MotionApps20.cpp.o
[... 223 lines cut ...]
Archiving .pio\build\esp32dev\libFrameworkArduino.a
Linking .pio\build\esp32dev\firmware.elf
Retrieving maximum program size .pio\build\esp32dev\firmware.elf
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  16.5% (used 54056 bytes from 327680 bytes)
Flash: [======    ]  55.7% (used 1095225 bytes from 1966080 bytes)
Building .pio\build\esp32dev\firmware.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 26 ELF sections
Successfully created esp32 image.
bin_copy([".pio\build\esp32dev\firmware.bin"], [".pio\build\esp32dev\firmware.elf"])
App Version: 0.0.1
App Name: Spot-Micro
Build Target: esp32dev
Renaming file to build\firmware\Spot-Micro_esp32dev_0-0-1.bin
Calculating MD5: ca747dbb10652e194ad9d114e67ca760
============================= [SUCCESS] Took 438.29 seconds =============================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       SUCCESS   00:07:18.292
============================== 1 succeeded in 00:07:18.292 ==============================
 *  Terminal will be reused by tasks, press any key to close it.

runeharlyk, 2024-07-26:
It seems the firmware usage is lower than I would have expected. Did you upload a filesystem image?

From the docs: "The filesystem image only has to be uploaded the first time and will override config files on the microcontroller." This means that you have to configure the network again after uploading a filesystem image.

As for the "URL not found" look in the file `WWWData.h` file to look if the entry are there. The file should be hundreds of lines

runeharlyk, 2024-08-18:
Closing the issue due to inactivity

closed 2024-08-18 by runeharlyk